Sleeper85/esphome-yambms #5: Is correct that CCL=0.0A

txubelaxu, 2024-11-04:
[image]

_Originally posted by @txubelaxu in https://github.com/Sleeper85/esphome-yambms/issues/4#issuecomment-2453557685_

Sleeper85, 2024-11-04:
Please read :

[Max current](https://github.com/Sleeper85/esphome-yambms/blob/main/documents/README/YamBMS_functions.md#max-current-)

[Requested values](https://github.com/Sleeper85/esphome-yambms/blob/main/documents/README/YamBMS_functions.md#requested-values)

[Auto CCL function](https://github.com/Sleeper85/esphome-yambms/blob/main/documents/README/YamBMS_functions.md#auto-ccl--dcl)

Be careful, the `Auto CCL` function can have an impact on what is requested from your inverter.
I advise you not to activate it to check that everything is working well.

Sleeper85, 2024-11-04:
Can you install the `YamBMS` dashboard and show me the part I have framed in pink?

[image]

txubelaxu, 2024-11-04:
This is with jk-bms-can version (is the version I have running now):
[image]

txubelaxu, 2024-11-04:
[image]

Sleeper85, 2024-11-04:
How are the settings below set?

[image]

[image]

Sleeper85, 2024-11-04:
This is an old version, please use `YamBMS 1.4.5`

txubelaxu, 2024-11-04:
Yes, I know. But it is the only version that does not restarts the system 1 time per 3 minutes.
The YamBMS 1.4.5 restarts each 3 minutes and the inverters detects low battery, inverters' relays start "jumping" and sometimes the powersource goes down.
Sorry, but the only stable for me is not the last one.

I would like to use the last one, of course. It is why I am debugging the code.

Sleeper85, 2024-11-04:
Ok then we will come back to this problem later.

txubelaxu, 2024-11-08:
Yes, it was because the settings were wrong: I had activated "auto" settings.

Deactivating them, CCL that appears in Victron console is correct.

Sleeper85, 2024-11-08:
Normally `Auto CCL` reduces the current to `0 A` only if `max_cell_v` reaches or exceeds the `OVPR` voltage you have configured in your BMS.

Maybe this shows an unbalanced battery or a very low `OVPR` voltage?

txubelaxu, 2024-11-08:
I have to inspect with more detail. 
But I have a OVPR= 3.448V and all the cells are about 3.315V on both BMSs. 
Delta is < 0.005V

Sleeper85, 2024-11-08:
> I have to inspect with more detail. But I have a OVPR= 3.448V and all the cells are about 3.315V on both BMSs. Delta is < 0.005V

Your OVPR is very low.

What is your `bulk_cell_v`?

I charge my cells at `3.45V` my OVPR is `3.55V` and my OVP `3.65V`.

MrPabloUK, 2024-11-08:
Yes, the logic for the AutoCVL will not work correctly if bulk voltage is higher than OVPR.
It's not something I would expect to see, especially if the OVP value is set to 3.65v.

In this case, I would increase the value of OVPR to something like 3.60v, or even 3.55v if preferred.

Sleeper85, 2024-11-08:
Will be added in the doc.

> [!IMPORTANT]  
> The `Auto CCL` function uses the `OVPR` value of your BMS which must be higher than `cell_bulk_voltage` to work properly. `e.g. for LFP` : CVL=3.45V / OVPR=3.55V / OVP=3.65V.
> The `Auto DCL` function uses the `UVPR` value of your BMS. `e.g. for LFP` : UVPR=3.00V

> [!IMPORTANT]  
> The `Auto CVL` function uses the `Balance Trig. Volt.` value of your BMS. `e.g. for LFP` : BTG=0.010V

clowrey, 2024-11-08:
Maybe the code could even send out an error if the value is higher? Users don't always read docs ;)

Sleeper85, 2024-11-08:
> Maybe the code could even send out an error if the value is higher? Users don't always read docs ;)

Good idea ;-)

Sleeper85, 2024-11-08:
> If I am not wrong, in the JK-PB BMSs: OVPR cannot be higher than Bulk Voltage.

Can you check it for us ?

MrPabloUK, 2024-11-08:
If OVPR can't be set higher than bulk, I can adjust the code to reference the mean of OVPR and OVP.
That should work ok for either type of JK BMS

txubelaxu, 2024-11-08:
> > If I am not wrong, in the JK-PB BMSs: OVPR cannot be higher than Bulk Voltage.
> 
> Can you check it for us ?

Confirmed: You can see here explained --> https://youtu.be/eZX7CkEEq_8?si=6OH4QK2l22zRvXqQ&t=442

So, OVP > RCV > SOC100% > OVPR > UVPR > SOC0% > UVP

Sleeper85, 2024-11-08:
Ok Thanks.

We will find another solution to do this.

@MrPabloUK is it possible to use the `balance_trigger_voltage` like you did for `Auto CVL`?

I'm trying to reduce the number of bms sensors needed (I added JDB and Seplos) and I removed `OVP` yesterday because it wasn't used in the code.

MrPabloUK, 2024-11-08:
Yes, we could use balance trigger voltage.
AutoCCL was originally designed to prevent cell voltage spikes at the end of charging, specifically to avoid hitting OVP.

There's no reason that the AutoCCL couldn't use `cell-bulk_voltage + (10 * balance_trigger_voltage)` as the upper threshold, for example.

In this case, that would be 3.45v + 0.10v = 3.55v (as standard).
The multiplier could be configurable, but 3.55v gives enough headroom for momentary cell voltage spikes.

Sleeper85, 2024-12-07:
Solved with YamBMS 1.5.1

closed 2024-12-07 by Sleeper85