Sleeper85/esphome-yambms #6: [JK-PB RS485] Error **Object is nullptr** with YamBMS and **bms_JK-PB_RS485_bms_minimal.yaml**

Sleeper85, 2024-11-07:
Hi @txubelaxu 

When we use the minimal version with less sensors, we get errors like below:

```python
[15:30:31][E][jk_rs485_bms:1625]: Object is nullptr
[15:30:31][E][jk_rs485_bms:1625]: Object is nullptr
[15:30:31][E][jk_rs485_bms:1625]: Object is nullptr
[15:30:31][E][jk_rs485_bms:1625]: Object is nullptr
[15:30:31][E][jk_rs485_bms:1625]: Object is nullptr
[15:30:31][E][jk_rs485_bms:1625]: Object is nullptr
[15:30:31][E][jk_rs485_bms:1625]: Object is nullptr
[15:30:31][E][jk_rs485_bms:1625]: Object is nullptr
```

[See comments on the DIY solar forum.](https://diysolarforum.com/threads/yambms-jk-bms-can-with-new-cut-off-charging-logic-open-source.79325/page-77#post-1192858)

Thanks to check that when you have time.

txubelaxu, 2024-11-07:
Hi @Sleeper85,

That error is not an error. Because the code try to update sensors/numbers... that are not really used. I have changed from W --> VV or from E -->VV

Could you please confirm that everything is OK now?

Many thanks!!

Sleeper85, 2024-11-08:
Thank you, I await feedback from users, I will keep you informed.

txubelaxu, 2024-11-08:
You are welcome!! ;-)

closed 2025-01-16 by Sleeper85