Sleeper85/esphome-yambms #7: Shunt Battery Capacity

goldserve, 2024-11-08:
If shunt is present, installed battery capacity should not be Shunt capacity + BMS capacity because that is assuming you enter the overall total capacity in the shunt config. If a BMS goes offline, maybe you can minus the capacity but not add.

Sleeper85, 2024-11-08:
Shunt capacity is not a parameter used in YamBMS.
Only these information are used: **Voltage, Current, Power and SoC**.

https://github.com/Sleeper85/esphome-yambms/blob/main/documents/README/YamBMS_behavior.md#shunt

goldserve, 2024-11-08:
I am confused where the 850Ah comes from when the only one BMS is reporting 270Ah then...

[image]

Sleeper85, 2024-11-08:
The `Installed Battery Capacity` value is the sum of the `Ah` of all your battery packs. At ESP32 startup this value is `0`, this value increases when a BMS is combined for the first time (only the first time) and it will never decrease after that (except after reboot).

In the example below, `3x 280Ah` batteries have been combined, then a BMS fails or you uncombine a BMS.
The `Installed Battery Capacity` value remains unchanged but the `Battery Capacity` value is decreased by 280Ah.

`Capacity Remaining` represents the Ah remaining in `Battery Capacity`.

[image]

goldserve, 2024-11-08:
I'm trying to track this down in code because I only have 1 BMS...Is it possible HA restored an old value and just kept on incrementing this because I switched ESPHome hardware but kept the same entities?

Sleeper85, 2024-11-08:
> I'm trying to track this down in code because I only have 1 BMS...Is it possible HA restored an old value and just kept on incrementing this because I switched ESPHome hardware but kept the same entities?

It is possible, you can completely delete your ESP32 in HA as well as all entities and then add it again.

**bms_combine.yaml**

```C++
            // Combine once only
            if (id(bms${bms_id}_combined_once) == false)
            {
              id(${yambms_id}_total_installed_battery_capacity) += id(bms${bms_id}_battery_capacity).state;
              id(bms${bms_id}_combined_once) = true;
            }
```

goldserve, 2024-11-08:
Not related to code but probably my Home Assistant. I saw the variable increment from 580 + 270 but i don't remember how I entered that 580 to begin with.

closed 2024-11-08 by goldserve

Sleeper85, 2024-11-08:
@goldserve 

Can you please check the message I sent you on the `DIY Solar` forum.

Thanks