Sleeper85/esphome-yambms #8: [ETH01-EVO POE] compilation problem with `DM9051` Ethernet component

Sleeper85, 2024-11-08:
### Discussed in https://github.com/Sleeper85/esphome-yambms/discussions/2

<div type='discussions-op-text'>

<sup>Originally posted by **Imanol82** November  2, 2024</sup>
Hello!

I'm making a new version of the yambms to put inside the box I printed in 3D. 
I have already soldered all the wires and when I start to compile the yambms.yaml that I have running to install it on this new board I get this error. 
What am I doing wrong?

[image]

Thank you very  much!!

</div>

Sleeper85, 2024-11-08:
https://github.com/esphome/esphome/pull/6861#issuecomment-2464611879

Imanol82, 2024-11-08:
Amazing. I've updated ESPHome and the firmware. Everything is working again.

Stupid question, why appears 1.4.4B as last instead of 1.4.5?

[image]

Thank you my friend, you're a master!!

Sleeper85, 2024-11-08:
It seems like you just didn't download the latest version from Github ?

Sleeper85, 2024-11-08:
> Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!

The default flash size for an `ESP32-C3-DevKitM-1` board is `4MB` but it seems your `ETH01-EVO` board only has 2MB of Flash?

Try this:

```YAML
esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  flash_size: 2MB
  framework:
    type: esp-idf
    version: latest
    platform_version: 6.5.0
```

Imanol82, 2024-11-08:
No. If I put this, I receibe this error:

```
Error: The program size (944880 bytes) is greater than maximum allowed (786432 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 181.36 seconds =========================
```

So I take out. with warning but is working...

Thank you very much!

Imanol82, 2024-11-09:
Hi Again.

Yesterday after last update... boooom! hahaha

Comunication between Deye, YamBMS and BMSs work fine, but I can't see nothing in HA! hahaha

[image]
(forget the middle area, I have been working on the installation and the battery has been disassembled)

[image]

[image]

How angry it is when things break due to updating... hahaha

Thanks my friend and sorry for being so annoying.

Sleeper85, 2025-04-13:
@Imanol82 

> Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!

I don't know why we have this warning because it seems this board has 32MB of Flash.

[image]

Imanol82, 2025-04-13:
32Mbit = 4MB

Imanol82, 2025-04-13:
@Sleeper85, you're an artist.

Everything working again on the ESP32 ETH01-EVO. And all with remote installation. It's perfect :)

[image]

I'm going to connect to the system an check if all is working fine.

One question, the shunt, is neccesary for something? I've 2 JK-PB and a Deye 6K.

Thank you very much my friend!!!!!

Sleeper85, 2025-04-13:
@Imanol82 

> One question, the shunt, is neccesary for something? I've 2 JK-PB and a Deye 6K.

It's well known that the JK-BMS SoC isn't very accurate. This isn't a problem in summer when you regularly charge your battery to 100%, but it becomes a problem in winter.

If you add a shunt, YamBMS will use the shunt's voltage, current, power, and SoC as a reference and transmit it to your inverter.

Imanol82, 2025-04-13:
I've solve this with an automation. When the diference between both SOC is higher than 5%, I charge fron the line this night until 100% hehe

Sleeper85, 2025-07-25:
Davicom DM9051 SPI Ethernet Controller is now integrated in esphome `2025.7`

closed 2025-07-25 by Sleeper85