Sleeper85/esphome-yambms #11: Auto CCL / DCL functions with JK-PB and new JK-B

Sleeper85, 2024-11-14:
@MrPabloUK

The JK-PB and new JK-B (JK-B2A8S20P) found on software `V11.54` must have `OVPR` and `UVPR` values ​​between the SOC 100% and SOC 0% values ​​which poses a problem with the Auto CCL and DCL functions which rely on these parameters.

[image]

MrPabloUK, 2024-11-14:
[https://github.com/Sleeper85/esphome-yambms/issues/5#issuecomment-2465844332](url)

I was originally planning to adjust the Auto CCL to work as per the comment above, although now I am thinking the below would be more consistent:
Auto CCL - use midpoint of OVPR and OVP (in this case, 3.55v with rounding).
Auto DCL - use midpoint of 0% SOC and UVP (in this case, 2.90v with rounding).

We can't use the 0% SOC or 100% SOC values, as not all BMSs will have that.

I can make that change pretty easily if you agree with the approach, @Sleeper85.

MrPabloUK, 2024-11-14:
The downside of setting DCL to 0 amps at UVPR is that it's possible for the BMS to still return a SOC >0% due to SOC drift.
Ideally, I'd like to ensure SOC is reset in that situation so that the user can't be confused about why the battery is no longer discharging.

I see your point about wanting to remove sensors where possible.
I do want to retain the use of balance_trigger_v where possible, as a user may prefer a 10mV range, or perhaps even more.
Setting it to a fixed 5mV would result in a very tight range for the PI controller to maintain.

If OVP has to be removed, I would prefer something like this for AutoCCL:
```
            double balance_delta_multiplier = id(${yambms_id}_balance_trigger_voltage).state * 10;
            double max_allowed_voltage = cell_bulk_v + balance_delta_multiplier;
```
Using a 10x multiplier would result in 3.549v as the max allowable voltage, giving a good range for the current to be tapered down smoothly.
The potential issue is that if someone sets OVP to 3.55v then there's a good chance BMS protection will kick in.

Sleeper85, 2024-11-14:
Otherwise replace `OVPR` and `UVPR` with **(OVP - offset_v)** and **(UVP + offset_v )** ?

I could also use **(UVP + offset_v )** for the **max discharge voltage** and not use `UVPR` anymore.

I connected to a French user via TeamViewer this afternoon because his **max charging current** was yo-yoing with `Auto CCL` enabled, here is his JK-PB config.

What kind of `offset_v` could we use ? It could be a fixed variable in the YAML.

[image]

MrPabloUK, 2024-11-14:
It's not a surprise that the requested CCL was bouncing around. With the current logic, having bulk above OVPR is like driving with your feet on the brake and accelerator at the same time.

What about replacing OVPR with ((bulk + OVP) /2)?
We still remove one sensor, plus it will prevent the oscillation of CCL as bulk is approached.
No need for a fixed offset that might clash with other settings.

UVPR could be replaced with (UVP + 0.1).
Alternatively, we could use UVP directly. I would just need to adjust the curve parameters to adjust the rate of current reduction.

Sleeper85, 2024-11-15:
It's OK for me 👌

I will modify the current version and provide access to the UVP and OVP sensor and then create a development branch for you to work on.

Sleeper85, 2024-11-15:
@MrPabloUK 

I fixed both **Auto CCL and DCL** functions as agreed.

* cell_high_v = (OVP + cell_bulk_v) / 2
* cell_low_v = UVP + 0.2

For **cell_low_v** I prefer 0.2V e.g. UVP 2.8 + 0.2 = 3V and many people use a UVP lower than 2.8V

MrPabloUK, 2024-11-15:
This works for me, I'll look into the CL curve tomorrow for alternatives if needed

Sleeper85, 2024-11-20:
Solved with YamBMS 1.5.1