Sleeper85/esphome-yambms #14: 'sensor.yambms_yambms_1_battery_capacity_s' has "unknown state"

vl-pub, 2024-12-24:
hello

'sensor.yambms_yambms_1_battery_capacity_s' has "unknown state" - see screenshot

I can't find such sensor in board code (board - ESP32-S3)

Could you please check?

Thanks Val.
[image]

Sleeper85, 2024-12-24:
How many BMS are combined ?

vl-pub, 2024-12-24:
Only one

vl-pub, 2024-12-24:
Hmmm...
It's interesting....

After switching "Combine" control On/Off several times something updated and "Battery Capacity" displays correct value:

[image]

Also I've found following scenario:

1. Set "Combine" to "ON"
2. Disconnect LAN cable from BMS
3. Reboot ESP board

[image]

4. Connect LAN cable to BMS - correct values are displayed:

[image]

5.  Set "Combine" to "OFF" - "Battery SOH", "Battery capacity", "Capacity Remaining" sensors display zero values:

[image]

6. Set "Combine" to "ON" - - correct values are displayed again:

[image]

7.  Set "Combine" to "OFF"
8. Reboot ESP board

9. Nothing changed:

[image]

vl-pub, 2024-12-24:
Seems "Combine" must be set to "ON" value for one connected BMS 
I'll check dashboard tomorrow morning one more time because of issue from topic has reproduced after some time after reboot.

thanks

vl-pub, 2024-12-25:
hello
here is update.

after several hors and switching between YamBMS/Control/BMS tabs "Battery Capacity" sensor lost value and has status "Unknown" ( "Combine" is set to "ON"):

[image]

"Combine" value has been changed between "ON"/"OFF" sever times - nothing changed

ESP board has been restarted, and "Combine" value has been changed between "ON"/"OFF" sever times again  - 

"Battery Capacity" sensor became available and has correct value:

[image]

So, some problem is still exist.
Could you please check from your side?
I'm ready to provide needed information / perform required testing.

Thank you,
Val.

Sleeper85, 2024-12-25:
Can you send me your YAML please?

vl-pub, 2024-12-25:
What yaml? Dashboard Yaml?

Sleeper85, 2024-12-25:
No, your main YAML.

vl-pub, 2024-12-25:
attached

[one-bms_JK-PB_RS485_mode2_MY.yaml.txt](https://github.com/user-attachments/files/18246330/one-bms_JK-PB_RS485_mode2_MY.yaml.txt)

Sleeper85, 2024-12-26:
> Seems "Combine" must be set to "ON" value for one connected BMS

An uncombined BMS is an unused BMS, all this is explained in the doc. You can uncombine it with the `Combine` switch and it can be uncombined automatically if the combination conditions are no longer respected.

Everything you have tested and shown in the picture is normal. If you have no combined BMS the values ​​are at 0% and the information is no longer sent on the CAN bus.

The only thing that is not normal is to have an `Unknown` value.

**Installed Battery Capacity:** this value is updated when a BMS is combined for the first time, then the value will remain unchanged if the BMS is uncombined, this corresponds to your total capacity regardless if the BMS are combined or not.

**Battery Capacity:** this is the sum of the battery capacity of all BMS currently combined. This value is updated when a BMS is combined or uncombined.

**Remainnig Capacity:** the remaining capacity of all BMS currently combined. This value is updated continuously.

vl-pub, 2024-12-26:
thank you for answer!

I think this issue can be closed now or after some time.
I'll inform you if "The only thing that is not normal is to have an Unknown value." issue occurred again and I'll have clear scenario.

Thanks again!
Val

Sleeper85, 2024-12-26:
Ok, I'll leave this issue open, keep me informed.

Sleeper85, 2025-01-16:
FYI

https://github.com/esphome/esphome/pull/7547

Sleeper85, 2025-01-18:
In YamBMS 1.5.3 I changed the condition to combine battery capacity info.

Can you tell me if it works better for you?

vl-pub, 2025-01-19:
is it ok that versions are differ??

[image]

Sleeper85, 2025-01-19:
Yes this is normal, I leave a few days to make corrections if necessary before creating the version tag.

vl-pub, 2025-01-19:
Hmmm...

It's a little bit worse now.

1. I unplug RS485 cable from BMS 
2. All sensors are in Unavailable state
3. Plug cable again
4. All sensors restored except "Battery Capacity" sensor
5. ESP32 reboot is required to restore normal work

Sleeper85, 2025-01-19:
And the `Installed Battery Capacity` too ?

vl-pub, 2025-01-19:
1. BMS unplugged

[image]

2. BMS plugged again

[image]

Sleeper85, 2025-01-19:
Ok I check it this afternoon.

Sleeper85, 2025-01-19:
Fixed.

Copy the new packages folder and recompile to test it.

vl-pub, 2025-01-19:
Looks like issue is fixed
Tested 2 scenarios mentioned below

Thanks
Val

-----------------------------------

**Scenario1**

1. Unplug BMS cable

[image]

2. Plug BMS cable

[image]

**Scenario 2**

1. Unplug BMS cable
2. Reboot ESP32

[image]

3. Plug BMS cable

[image]

Sleeper85, 2025-01-19:
I just made another correction => [Don't count a uncombined BMS as blocking charge/discharge](https://github.com/Sleeper85/esphome-yambms/commit/91df96d70cdfe5b9f13691cc72330e9b5c1dcecd)

This is unrelated to this issue.

closed 2025-01-19 by Sleeper85