Sleeper85/esphome-yambms #16: Poor performance of ESP32 Devkit V1

VolodymyrPavlyuk, 2025-01-10:
I have a Victron SmartShunt and three JK-PB BMSs and this is the maximum configuration which can run on my Devkit V1:
```yaml
packages:
  device_base: !include packages/base/device_base.yaml
  device_board: !include packages/board/board_esp32-devkit-v1.yaml

  # Config : this YAML contains the settings shared between all ESP32s
  config_yambms: !include packages/config/config_yambms.yaml

  shunt1: !include
    file: packages/shunt/shunt_combine_Victron_SmartShunt_BLE.yaml #
    vars:
      shunt_id: '1' # must be a number
      shunt_name: 'Shunt'
      shunt_mac_address: 'c9adfc3f5dc2'
      shunt_encryption_key: '908b0661a62f36c026424b21080fa4ae'

  # The sniffer represents the ESP node connected to the JK-BMS PB RS485 bus
  # If address 0x00 is not detected on the RS485 bus, the sniffer will act directly as a master BMS (mode 2, max 15 BMS)

  sniffer1: !include
    file: packages/bms/bms_combine_JK-PB_RS485_sniffer.yaml
    vars:
      sniffer_id: 'sniffer1'
      sniffer_protocol: 'JK02_32S'
      sniffer_uart_id: 'uart_esp_1'

  # Mode2 : configure the DIP switches of your BMS from 0x01 to 0x0F (don't use the 0x00 address, maximum 15 BMS)
  bms1: !include
    file: packages/bms/bms_combine_JK-PB_RS485_bms_minimal.yaml
    vars:
      # Sniffer ID
      sniffer_id: 'sniffer1'
      # BMS vars
      bms_id: '1' # must be a number
      bms_name: 'JK-BMS 1'
      bms_rs485_address: '0x01' # BMS 1 DIP switch
  
  # bms2: !include
  #   file: packages/bms/bms_combine_JK-PB_RS485_bms_minimal.yaml
  #   vars:
  #     # Sniffer ID
  #     sniffer_id: 'sniffer1'
  #     # BMS vars
  #     bms_id: '2' # must be a number
  #     bms_name: 'JK-BMS 2'
  #     bms_rs485_address: '0x02' # BMS 2 DIP switch

  # bms3: !include
  #   file: packages/bms/bms_combine_JK-PB_RS485_bms_minimal.yaml
  #   vars:
  #     # Sniffer ID
  #     sniffer_id: 'sniffer1'
  #     # BMS vars
  #     bms_id: '3' # must be a number
  #     bms_name: 'JK-BMS 3'
  #     bms_rs485_address: '0x03' # BMS 3 DIP switch
```
Even with this config esp resets every minute or so and the reason is `Software Reset CPU`:
[image]
[image]

When I uncomment the second BMS the board resets 5 times and then just freezes.

Is this expected behaviour for such board or that's a sign that my board is defective?

Sleeper85, 2025-01-10:
[image]

[BLE components are resource hungry and can lead to this kind of problems](https://esphome.io/components/ble_client.html).

Then a classic ESP32 is always less powerful than an ESP32-S3 but, without being sure, I think that the quality of the boards / chipsets can be different between known brands and low-cost boards.

For example, I know that an AtomS3 Lite (ESP32-S3) is able to supervise 3x BLE BMS and with an ESP32-S3 DevKitC-1 board I can't do it.

Why? good question... are there different quality of chipsets? I don't know the answer at the moment...

It's a good idea to use `bms_combine_JK-PB_RS485_bms_minimal.yaml` as in your example.

I think that if you remove the BLE Shunt you should be able to supervise your 3x JK-PB.

clowrey, 2025-01-10:
Maybe just better antenna on the M5Stack product results in less retries on the Bluetooth. Chipset should be the same eh?

Sleeper85, 2025-01-10:
There are several ESP32-S3 versions possible so if an `AtomS3` is able to supervise 3x BMS BLE I don't understand why an `ESP32-S3 N16R8` couldn't do it.

But what is certain is that the BLE stack consumes a lot of resources and sometimes causes problems when adding other components.

I'm currently doing new tests with BLE...

[image]

Sleeper85, 2025-01-10:
Good news, I confirm that I managed to make YamBMS work with 3x BMS BLE on an ESP32-S3.

Against all expectations, the activation of the PSRAM was the cause of the problem with as a result an overconsumption of DRAM and an instability/crash.

That is why it works well on an AtomS3 (it does not have PSRAM).

The PSRAM will be disabled by default in the next version.

VolodymyrPavlyuk, 2025-01-10:
Another good news, I confirm that the original issues was caused by BLE.
I connected the shunt via UART and that took me few minutes because I already had the cable, and now even Devkit V1 handles a shunt and three JK-PB BMSs without any performance issues. BMSs use minimal configuration for now and honestly I don't see a reason to change it.

@Sleeper85, thanks for help!

VolodymyrPavlyuk, 2025-01-13:
It's been few days with disabled BLE and my esp devkit v1 has been working flawlessly. I don't know yet what I'm going to do with new S3 boards which still are on their way. Maybe for future proofing I'll migrate to S3, but that's definitely isn't a priority because everything works fine on v1 and don't wan't fix what isn't broken.

I've disabled the AP and I don't see any noticeable difference, but hope that less unused code actually increases stability.

It the and I'm very happy with the project because the charging logic implemented in it is how to should work natively, but for some reason it doesn't. And in combination with Victron SmartShunt the calculation of SoC is way more accurate than only with JK PB BMSs.

closed 2025-01-16 by Sleeper85