Sleeper85/esphome-yambms #20: JBD RS485

andrewfraley, 2025-01-16:
I have a JBD with a two wire RS485 connector so I'd like to use RS485 and retain the use of the UART connector for the bluetooth module.  I don't see support for JBD R485, but I'd like to help add it.  Before I head down a dead end, has anyone tried this before?

I have an [RS485 to TTL module](https://www.amazon.com/dp/B0B9XT9HM8?ref=ppx_yo2ov_dt_b_fed_asin_title), and an [Adafruit S3 Feather](https://www.adafruit.com/product/5691), should these be adequate if combined with a compatible CAN transceiver?

Sleeper85, 2025-01-16:
I have never had a `JBD`, this is something you need to test using the `modbus` and `modbus controller` components provided with esphome.

If you can get the BMS infos from RS485, I could add it to YamBMS afterwards.

The hardware you have is OK but it will be necessary to create a `board_esp32-s3_Adafruit_S3_Feather.yaml` file adapted to your board.

andrewfraley, 2025-02-06:
I didn't realize this would be as simple as it was.  The JBD uses the same protocol for all connections so the LilyGo T-CAN485 works just by connecting its RS485 pins to the RS485 pins on the BMS and then using `packages/bms/bms_combine_JBD_UART_full.yaml` for the BMS config.  

The only thing I changed was setting the UART baud rate to 9600 in `packages/board/board_ESP32_LilyGo-T-CAN485.yaml`

closed 2025-02-06 by andrewfraley