Sleeper85/esphome-yambms #21: multiple JBD BT bms to RS485

uploader18, 2025-01-18:
Hello!

Will there be an option to read data from 3 JBD bms's over bluetooth and send data to inverter over RS485 battery port?

Regards Žiga

Sleeper85, 2025-01-18:
Getting the data from the JBD BMS from Bluetooth will not be very complicated, however for the moment there are only CAN protocols supported for the inverter communication.

The inverter communication protocol over RS485 should be developed and tested and this part will take more time.

uploader18, 2025-01-18:
If I buy one SMG II 6.2 KW connect esp32(with esphome and web server) on battery rs485 port and give you access to it if you are willing to take a shoot at it. And a JK bms(sice i have it) with bluetooth connected to battery.

Sleeper85, 2025-01-19:
You are the second one who asks me this week for SMG II. The PYLON RS485 protocol is not something standard that is easy to program, it can take a lot of time in programming/testing/debugging.

It would be a good idea to do it for inverters that only support RS485 communication but I don't have time at the moment. I have other projects to do at the beginning of the year. The day I decide to do it I could test it with my Deye inverter.

https://github.com/Sleeper85/esphome-yambms/issues/17

closed 2025-01-29 by Sleeper85