Sleeper85/esphome-yambms #22: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"

vl-pub, 2025-01-19:
Hello,
I understand that this ticked is not real issue,
But I guess you have take into account on this warning reproduced during firmware compilation

Thanks,
Val

...............

Compiling .pioenvs/yambms/src/esphome/components/light/esp_color_correction.cpp.o
Compiling .pioenvs/yambms/src/esphome/components/light/esp_hsv_color.cpp.o
**In file included from src/esphome/components/esp32_rmt_led_strip/led_strip.h:12,
                 from src/esphome/components/esp32_rmt_led_strip/led_strip.cpp:2:
/home/vlak/.platformio/packages/framework-espidf/components/driver/deprecated/driver/rmt.h:18:2: warning: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h" [-Wcpp]
   18 | #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"
      |  ^~~~~~~**
Compiling .pioenvs/yambms/src/esphome/components/light/esp_range_view.cpp.o
Compiling .pioenvs/yambms/src/esphome/components/light/light_call.cpp.o

Sleeper85, 2025-01-19:
You will find more information in [this issue](https://github.com/esphome/issues/issues/5357).

You should know that there has been a big change since esphome `2024.12.0`, it is the replacement of ESP-IDF 4 by 5.

This error is therefore a warning and the author of the `esp32_rmt_led_strip` component will have to resolve it before it becomes blocking.

The introduction of ESP-IDF 5.1.5 may possibly cause problems for many components.

For this reason I continue to use esphome `2024.11.3` for the moment.

* esphome <= 2024.11.3 (ESP-IDF 4.4.8)
* esphome >= 2024.12.0 (ESP-IDF 5.1.5)

For my part, I can't do anything more.

vl-pub, 2025-01-19:
Ok,
thank you for answer!
You may close this issue if decide.

thanks
Val

closed 2025-01-19 by Sleeper85