Sleeper85/esphome-yambms #23: Multi JK-PB x2 one BMS going offline every 3 days

maks2204, 2025-01-22:
I have 2 BMS in parallel, but BMS with address 0x01 disable every 3 days at 7 am +- 5 min. 
What could be the problem?
I am using RS485 isolated board (high speed dual)

```
# Updated : 2024.10.03
# Version : 1.4.5
# GitHub  : https://github.com/Sleeper85/esphome-yambms

# YamBMS ( Yet another multi-BMS Merging Solution )

# This YAML is free software: you can redistribute it and/or
# modify it under the terms of the GNU General Public License
# as published by the Free Software Foundation, either version 3
# of the License, or (at your option) any later version.
#
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
# See the GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with this program. If not, see <http://www.gnu.org/licenses/gpl.html>.

substitutions:
  name: yambms

# Don't forget to configure your WiFi credentials in the secrets.yaml file
#
# If needed, configure a static IP here
# wifi:
#   manual_ip:
#     static_ip: 192.168.0.85
#     gateway: 192.168.0.1
#     subnet: 255.255.255.0
#     dns1: 8.8.8.8
#     dns2: 8.8.4.4

logger:
  level: INFO # DEBUG / INFO / WARN

ota:
  platform: esphome

# Please use the native `api` component instead of the `mqtt` section.
# If you use Home Assistant, the native API is more lightweight.
# If there is no HA server connected to this API, the ESP32 reboots every 15 minutes to try to resolve the problem.
# If you don't use Home Assistant please uncomment the "reboot_timeout: 0s" option.
api:
  reboot_timeout: 15min

# If you don't want to use ESPHome's native API you can use MQTT instead.
# In this case don't forget to remove the 'api:' section.
# mqtt:
#  broker: !secret mqtt_host
#  username: !secret mqtt_username
#  password: !secret mqtt_password
#  id: mqtt_client

# Please note that enabling this component will take up a lot of memory and may decrease
# stability and be the cause of reboot depending on the capabilities of the board used.
#web_server:
#  port: 80
#  log: false
#  ota: false

# +--------------------------------------+
# | Packages                             |
# +--------------------------------------+
packages:
  #device_base: !include packages/base/device_base.yaml
  # Board : uncomment your board
  # device_board: !include packages/board/board_esp32-devkit-v1.yaml
  #device_board: !include packages/board/board_esp32-s3-devkitc-1.yaml
  # device_board: !include packages/board/board_atom-s3-lite.yaml
  device_board: !include packages/board/board_ESP32-S3_DevKitC-1.yaml
  
  # device_board: !include
  #   file: packages/board/board_atom-s3-display.yaml
  #   vars:
  #     yambms_id: 'yambms1'
  #     canbus_id: 'canbus1'

  # Config : this YAML contains the settings shared between all ESP32s
  config_yambms: !include packages/config/config_yambms.yaml

  # shunt1: !include
  #   file: packages/shunt/shunt_combine_Victron_SmartShunt_UART.yaml # shunt_combine_Junctek_KHF_UART.yaml
  #   vars:
  #     shunt_id: '1' # must be a number
  #     shunt_name: 'Shunt 1'
  #     shunt_uart_id: uart_esp_0

  # The sniffer represents the ESP node connected to the JK-BMS PB RS485 bus
  # If address 0x00 is not detected on the RS485 bus, the sniffer will act directly as a master BMS (mode 2, max 15 BMS)

  # sniffer1: !include
  #   file: packages/bms/bms_JK-PB_RS485_sniffer_talk_pin.yaml
  #   vars:
  #     sniffer_id: 'sniffer1'
  #     sniffer_protocol: 'JK02_32S'
  #     sniffer_uart_id: 'uart_esp_1'
  #     sniffer_talk_pin: 8 # ESP32: 4, ESP32-S3: 8, ESP32-C3: 10

  sniffer1: !include
    file: packages/bms/bms_combine_JK-PB_RS485_sniffer.yaml
    vars:
      sniffer_id: 'sniffer1'
      sniffer_protocol: 'JK02_32S'
      sniffer_uart_id: 'uart_esp_1'

  # Mode2 : configure the DIP switches of your BMS from 0x01 to 0x0F (don't use the 0x00 address, maximum 15 BMS)
  bms1: !include
    file: packages/bms/bms_combine_JK-PB_RS485_bms_full.yaml
    vars:
      # Sniffer ID
      sniffer_id: 'sniffer1'
      # BMS vars
      bms_id: '1' # must be a number
      bms_name: 'JK-BMS 1'
      bms_address: '0x01' # BMS 1 DIP switch

  bms2: !include
    file: packages/bms/bms_combine_JK-PB_RS485_bms_full.yaml
    vars:
      # Sniffer ID
      sniffer_id: 'sniffer1'
      # BMS vars
      bms_id: '2' # must be a number
      bms_name: 'JK-BMS 2'
      bms_address: '0x02' # BMS 2 DIP switch

  yambms1: !include
    file: packages/yambms/yambms.yaml
    vars:
      # Please read the cut-off charging logic README to understand how the YamBMS works
      yambms_update_interval: '1s'
      # Input numbers can be displayed as a slider or an input box, options are 'slider' or 'box'.
      yambms_input_number_mode: 'slider'
      # Please check and fill in the options below correctly according to your battery chemistry and number of cells in series.
      # These parameters are important and used for charging logic.
      # Battery Chemistry
      yambms_battery_chemistry: '1' # 1-LFP | 2-Li-ion | 3-LTO
      # Number of cells in series
      yambms_cell_count: '16'
      # Bulk / Absorption Voltage : corresponds to the Bulk voltage that will be used to charge the battery. (LFP : 55.2V = 3.45V/Cell for 16S battery)
      yambms_bulk_v: '55.2'
      # Float Voltage : corresponds to the voltage at which the battery would be maintained at the end of charge. (LFP : 53.6V = 3.35V/Cell for 16S battery)
      yambms_float_v: '53.6'
      # Rebulk voltage, voltage from which a new Bulk charge can start. (LFP : 52.8V = 3.3V/Cell for 16S battery)
      yambms_rebulk_v: '52.8'
      # Maximum time in minutes that the cut-off step can last before charging is complete
      # If your cells are properly balanced this step ends at the fastest after the `cut-off timer`
      # This timer can be deactivated with a switch
      yambms_eoc_timer: '30'
      # Time in seconds during which the end of charge conditions must be respected (cut-off + cells not equalizing)
      yambms_cutoff_timer: '60'
      # Maximum charging cycles is used to calculate the battey SOH, LF280K v3 =8000.0, LF280K v2 =6000.0, LF280=3000.0 (decimal is required)
      yambms_max_cycles: '6000.0'

  canbus1: !include
    file: packages/yambms/yambms_canbus.yaml
    vars:
      # CANBUS vars
      canbus_id: 'canbus1'
      canbus_name: 'CANBUS 1'
      canbus_node_id: 'canbus_node1'
      canbus_light_id: 'esp_light'
      # The CANBUS link will be considered down if no response from the inverter (ID 0x305) for 5s
      # The Deye inverter sends the ACK (can_id 0x305) only when it receives the can_id 0x356
      canbus_link_timer: '5s'

# +--------------------------------------+
# | DEBUG ( logger level must be DEBUG ) |
# +--------------------------------------+

  device_debug: !include
    file: packages/base/device_debug_ESP32.yaml
    vars:
      debug_name: 'Debug'
      debug_update_interval: '5s'

```

[image]
[image]
[image]
[image]
[image]

Sleeper85, 2025-01-29:
Strange...

Do you know that YamBMS version is `1.5.4` now ?

A lot of things have evolved since `1.4.5` including the component for JK-PB over RS485.

What is the firmware version of your JK-PB BMS ?

maks2204, 2025-01-29:
I'm using the latest version of YamBMS. I'm just editing an old config file. The comments are left from old file.
BMS version 15.35.
Currently it has been working flawlessly for almost 7 days. Maybe the latest update helped.

[image]

maks2204, 2025-02-01:
Same problem after 3 days of uptime.
[image]
[image]
[image]

Sleeper85, 2025-04-13:
Do you still have this problem?

txubelaxu, 2025-04-13:
I have a problem that esp32 reports "can bus offline" after some days. CAN
bus is not recovered. So, I hace to restart the esp32 to recover the
communication. I have made a rule in HA.

Could the yambms code to reconnect automatically/periodicallyCANbus when
disconnected?

Many Thanks

Txubelaxu

Sleeper85, 2025-04-13:
@txubelaxu 

> Could the yambms code to reconnect automatically/periodicallyCANbus when
disconnected?

This is already the case, if YamBMS does not receive an `ACK 0x305` from your inverter, the CAN bus link will pause for `60s` before attempting to communicate again every 60s.

With the default settings, your inverter has `5s` to send an ACK, and it should normally send an ACK every `1s` according to the CAN protocol specification (PYLON).

[More information is available in the documentation.](https://github.com/Sleeper85/esphome-yambms/blob/main/documents/README/YamBMS_behavior.md#can-bus-link)

You can also monitor your [inverter heartbeat](https://github.com/Sleeper85/esphome-yambms/blob/main/documents/README/YamBMS_functions.md#inverter-heartbeat-monitoring).

Have you experienced this problem often?

This has never happened in my installation.

maks2204, 2025-04-14:
71d uptime. Everything is fine
[image]

Sleeper85, 2025-04-14:
Good to hear, I close this issue.

closed 2025-04-14 by Sleeper85