Sleeper85/esphome-yambms #24: SoC never reaches 100%

volopav, 2025-01-27:
I have 3 JK inverter BMSs and a Victron SmartShunt and in general everything works fine, however SoC which is reported to the inverter via CAN never reaches 100%. Typically it's 98% and some times I can see 99%.

From physics standpoint the battery reaches 100% according to cut-off conditions:
[image]

So, voltage is correct, current is correct, transition from Bulk to Float is correct, however reported SoC isn't correct.

Here's my YamBms config:
```yaml
yambms1: !include
    file: packages/yambms/yambms.yaml
    vars:
      # Please read the cut-off charging logic README to understand how the YamBMS works
      yambms_update_interval: '1s'
      # Input numbers can be displayed as a slider or an input box, options are 'slider' or 'box'.
      yambms_input_number_mode: 'slider'
      # Please check and fill in the options below correctly according to your battery chemistry and number of cells in series.
      # These parameters are important and used for charging logic.
      # Battery Chemistry
      yambms_battery_chemistry: '1' # 1-LFP | 2-Li-ion | 3-LTO
      # Number of cells in series
      yambms_cell_count: '16'
      # Bulk / Absorption Voltage : corresponds to the Bulk voltage that will be used to charge the battery. (LFP : 55.2V = 3.45V/Cell for 16S battery)
      yambms_bulk_v: '56.2'
      # Float Voltage : corresponds to the voltage at which the battery would be maintained at the end of charge. (LFP : 53.6V = 3.35V/Cell for 16S battery)
      yambms_float_v: '53.6'
      # Rebulk voltage, voltage from which a new Bulk charge can start. (LFP : 52.8V = 3.3V/Cell for 16S battery)
      yambms_rebulk_v: '52.8'
      # Maximum time in minutes that the cut-off step can last before charging is complete
      # If your cells are properly balanced this step ends at the fastest after the `cut-off timer`
      # This timer can be deactivated with a switch
      yambms_eoc_timer: '20'
      # Time in seconds during which the end of charge conditions must be respected (cut-off + cells not equalizing)
      yambms_cutoff_timer: '60'
      # Maximum charging cycles is used to calculate the battey SOH, LF280K v3 =8000.0, LF280K v2 =6000.0, LF280=3000.0 (decimal is required)
      yambms_max_cycles: '6000.0'
```

Sleeper85, 2025-01-27:
Hi,

It seems that your BMS does not reach 100% itself.

The current code blocks the SoC at 98% max during the charging phase (useful for some inverters) then when switching to `EOC/Float` it is the average SoC of all BMS that is sent. If your BMS are not at 100% then it will not be 100%.

Sleeper85, 2025-01-27:
Ok I understand the problem, since version `15.35` JK has implemented a new charging logic to reach 100%. This charging logic is based on an `absorption timer`. It is a very simple charging logic that can overcharge your battery if it is poorly set.

YamBMS uses another more advanced charging logic based on `Cut-Off Voltage and Current`. If the Cut-Off conditions are met and your cells are equalized then the charge stops, this can go faster than a 60min absorption timer. The Cut-Off Charging Logic also has an `EOC Timer` enabled by default which is actually a safety that stops the charge in all cases after 30min (by default) in the `Cut-Off` phase. A well equalized battery can finish before 30min.

This `EOC Timer` switch can also be disabled, in this case the end of charge can last longer than 30min and will stop only if all your cells are equalized. You can try increasing the time of this timer or disabling it.

YamBMS used an `absorption timer` in the past and it has been replaced by this new `Cut-Off Charging Logic`.

I could also send `100%` at the end of the charge even if your JK-PB BMS still shows `99%`.

txubelaxu, 2025-01-27:
So, there are 2 timers: one hard timer (BMS) and one soft timer (yambms). If “absortion time” is present, I would disable the soft timer automatically at boot time, and assign a value of 2 x “absortion time”. If a user wants to have this second soft timer, the user could enable it manually.

In our case, BMSs set their SOC to 100% when some conditions meet. This [video at 12:40 and 14:00](https://youtu.be/7vEtdS8F9k0?si=vFXuGye1EQqRkEUh) shows the algorithm used by JK. I think it is very good explained here.

I would not send 100% even if JK-PB still shows 99%, because it is not real.

Sleeper85, 2025-01-27:
YamBMS uses its own charging logic (well before JK 15.35) which is in my opinion better than the JK inverter BMS so I don't see why it is necessary for YamBMS to follow the JK logic.

JK uses its own logic to determine the 100% SoC based on an `absorption timer`.

YamBMS uses its own logic (based on manufacturer recommendations) to determine the 100% SoC based on a `Cut-Off Voltage & Current`.

These are just two different logics and two ways to determine the 100% SoC.

If you want to use a simple logic based on an absorption timer then you can connect your JK master directly to your inverter without using YamBMS, right ?

If your JK-PB displays 99% when the YamBMS charge ends it is because the YamBMS logic does not overcharge the cells and is based on a safer logic than a simple absorption timer.

I know the new JK charging logic and that is all it takes to overcharge the cells.

Charging at `3.5V` until you get a charging current of 0A is already a form of overcharging according to the manufacturers' recommendations.

**JK logic**

[image]

**YamBMS logic**

[image]

volopav, 2025-01-28:
> It seems that your BMS does not reach 100% itself.
> 
> The current code blocks the SoC at 98% max during the charging phase (useful for some inverters) then when switching to `EOC/Float` it is the average SoC of all BMS that is sent. If your BMS are not at 100% then it will not be 100%.

Fortunately I gather a lot of data and can perform analysis retrospectively and indeed my BMSs don't reach 100% and neither does Victron shunt. I thought that YamBMS sets 100% SoC regardless of reading from BMS and/or a shunt, but thanks to your explanations it's not the case. So basically YamBMS cares about physical state of the battery cells regardless of how that's reflected in the reported SoC values. YamBMS still to do it's best, but it obviously doesn't have any capabilities to calculate SoC, to it has to rely on BMS or shunt. 

I really like the cutt-off logic of yambms because this is how LFP batteries should be changed, and I still wonder why BMS manufacturers cannot implement that in their software.

I decreased `Charged detection time` of the shunt to 2min and increased `yambms_cutoff_timer` to 300s and now I can see 100% for brief moment of time which is what I wanted. My cells are very nicely balanced, so they stay on Bulk voltage only for few minutes:
[image]

I'll try to decrease `yambms_cutoff_timer` to the smallest value which still allows my shunt to reset to 100%.

AlexanderSch90, 2025-03-28:
I wonder whether yambms can be operated sensibly with firmware >=15.35, as from then on you actually have a double, but different, loading logic. The question then arises as to whether it would make more sense to do without yambms or use a firmware <15.35. Do you have an opinion on this?

Sleeper85, 2025-03-31:
> I wonder whether yambms can be operated sensibly with firmware >=15.35, as from then on you actually have a double, but different, loading logic. The question then arises as to whether it would make more sense to do without yambms or use a firmware <15.35. Do you have an opinion on this?

If you use YamBMS, the internal JK logic is not used.
I don't think there's any point in downgrading.

What's the benefit of using YamBMS instead of JK logic alone?

1) YamBMS logic is smarter and won't overload your cells.
2) YamBMS allows you to add one or more shunts as a source for voltage, current, power, and SoC (more accurate than JK)
3) YamBMS provides other useful features like `Request Force Charge`, etc.

But if you wish, you can simply use part of the code without YamBMS to monitor your BMS and use the JK logic and CAN bus.

You're free ;-)

Sleeper85, 2025-04-03:
I didn't specify it but there is a security enabled by default with a `30min` timer (this time can be adjusted). Maybe you should disable this timer, this will give your battery more time at the `Bulk` charge voltage to equalize your cells and will also give more time to the JK logic to go to 100%.

[image]

The `EOC timer enabled` switch ensures that the `Cut-Off` phase lasts maximum `30 min` (default value) even if your cells are still being equalized. If your cells are equalized, the `Cut-Off` phase will end at the earliest after `60s` (default value). This prevents the risk of staying in the `Cut-Off` phase for many hours if you have several batteries and they are poorly equalized.

> [!IMPORTANT]
> If you have **totally unbalanced batteries** and want to fix this problem, you can disable the `EOC timer` and let `YamBMS` balance your batteries, **this can take hours**. A well balanced battery does not need this timer enabled as the equalization phase will finish in less than 30 minutes.

Sn0w82, 2025-06-03:
> > It seems that your BMS does not reach 100% itself.
> > The current code blocks the SoC at 98% max during the charging phase (useful for some inverters) then when switching to `EOC/Float` it is the average SoC of all BMS that is sent. If your BMS are not at 100% then it will not be 100%.
> 
> Fortunately I gather a lot of data and can perform analysis retrospectively and indeed my BMSs don't reach 100% and neither does the Victron shunt. I thought that YamBMS sets 100% SoC regardless of readings from a BMS and/or a shunt, but thanks to your explanations it's not the case. So basically YamBMS cares about physical state of the battery cells regardless of how that's reflected in the reported SoC values. YamBMS still tries to do it's best, but it obviously doesn't have any capabilities to calculate SoC, so it has to rely on BMS or shunt.
> 
> I really like the cutt-off logic of yambms because this is how LFP batteries should be changed, and I still wonder why BMS manufacturers cannot implement that in their software.
> 
> I decreased `Charged detection time` of the shunt to 2min and increased `yambms_cutoff_timer` to 300s and now I can see 100% for brief moment of time which is what I wanted. My cells are very nicely balanced, so they stay on Bulk voltage only for few minutes: <img alt="Image" width="2000" src="https://private-user-images.githubusercontent.com/5807389/407273484-01e8a955-9a9d-4e72-953d-d525aecbcb66.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDg5NDQzNzksIm5iZiI6MTc0ODk0NDA3OSwicGF0aCI6Ii81ODA3Mzg5LzQwNzI3MzQ4NC0wMWU4YTk1NS05YTlkLTRlNzItOTUzZC1kNTI1YWVjYmNiNjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDYwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA2MDNUMDk0NzU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGRkMzNiNDY1ZGY1YWE1YTE3NjAyMWUwNzVhMzFlMDcyNjBlZjcyZjhjMmE5ZmEyMjFmZGE2NGRmNzZhMTE4YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.bEyPGQzaTBzARgykSfzB8rokSQpiPL9amLso_p_-PfA">
> 
> I'll try to decrease `yambms_cutoff_timer` to the smallest value which still allows my shunt to reset to 100%.

what do you use to display data in such graphs? Grafana

volopav, 2025-06-03:
> what do you use to display data in such graphs? Grafana

Yes, it's Grafana.
I've connected InfluxDB to my Home Assistant and use Grafana for visualization.

Sleeper85, 2025-07-24:
The `RCV`, `RFV`, and `RCV timer` values are not used by YamBMS.
But they are used by the JK to reset the SOC to 100%.
If you are instructing YamBMS to charge at `3.5V/cell`, set the JK RCV value lower to `3.45V`.
Reduce the `RCV timer` to 15 minutes.
This will reset the SOC to `100%` faster.

Imanol82, 2025-07-24:
Noted! I'll try tomorrow.

Thank you very much my friend!!