Sleeper85/esphome-yambms #25: Why is "Broadcasting JK-PB settingts to all BMS" set to OFF by default

txubelaxu, 2025-01-27:
In 1.5.3 version, this change has been made. 

I think "by default", in general, is better to send the same change to all BMSs working in parallel. And it you need to send a particular setting for one of them, then switch to OFF manually previous to making the change.

But sure there will be a reason to switch the "default" value to OFF at the boot time. What is this reason?

Many Thanks!!

txubelaxu, 2025-01-27:
Well, I think it is more common to want to switch on/off charge/discharge to every BMS at the same time automatically. With more than one BMS you could reach easily charge/discharge intensity limits when switching off, for example, the discharge in a BMS and not in the other one.

For voltage params, security params, I think it is better to broadcast ON.
I have it switched ON always, and if I need to operate with only one BMS, then I switched it OFF before changing that parameter, and I hcange it to ON again.

But it is only my opinion.

Sleeper85, 2025-01-27:
You are probably right but users are confused and wonder why it behaves like that thinking it is a bug.
So it's important that they read the documentation to understand the behavior of this switch.

When it's activated, its state is memorized and it will remain activated even after a reboot and even after an upgrade.

txubelaxu, 2025-01-27:
understood

closed 2025-01-27 by txubelaxu