Sleeper85/esphome-yambms #32: Deye

sloniu82, 2025-02-28:
I'm using Deye hybrid system:
- Deye sun-12k-sg04lp3-eu
- Deye Battery RW-M6.1-B
From what I found in manual, it communicates with external components using CAN or RS485.
Unfortunately, I can't find any documentation with communication procotol data description.
Also, it seems, that Deye is not OEM'ing any of well known BMS'es in their batteries (or using custom firmwares).

Luckly, I found a project on github, where most of CAN messages are already decoded.
https://github.com/Adminius/esphome-yaml-collection/blob/main/deye_rw-m6.1.yaml

I wonder, is it possible to add YAMBMS support for Deye proprietary batteries.

It would be nice to have ability to add other brands or DIY BMS'es (potentially much cheaper! ) to the system.

Sleeper85, 2025-03-09:
Sorry for the late reply.

This battery probably uses the Deye PCS CAN protocol.
I have already developed this protocol for the output to the inverter (unpublished aplha).

It would probably be possible to connect the battery as a YamBMS input but this requires time for development and I am busy on other projects at the moment.

GHswitt, 2025-03-20:
Hi,

as I've also two Deye modules and want to connect them I could help out. However I first want to connect the main JK BMS battery packs and Smartshunts with YamBMS. If they are up and running I will look into the Deye packs.

Sleeper85, 2025-03-21:
I could provide the information I currently have regarding the protocols used by Deye batteries.

GHswitt, 2025-03-21:
Sure, I've also collected some about the CAN messages on both ports (PCS and IN/OUT). I would also like to use the interconnection CAN bus messages between the modules and not only the port to the inverter. With them it seems possible to get all cell voltages.

Sleeper85, 2025-03-21:
If you have the time to develop and test it, you're welcome to join.

I can guide you on the best way to integrate it into YamBMS.

GHswitt, 2025-03-21:
Thank you. I propose that I come back to you when I've sucessfully decoded the CAN messages via ESPHome, I've already got some configuration files, but need some time to setup the hardware besides other projects. Then for creating a packages/bms/bms_.. file I would appreciate the help.

GHswitt, 2025-03-28:
I want to share some results. Looks good so far. Still many things to do, but it does something. Here is my working branch: https://github.com/GHswitt/esphome-yambms/commit/cdbcce9fe3b0d08700b98fb355ce3e3a91855eed

In this setup I'm currently only using one CAN port connected to the PCS port of the battery. The battery has two modules.
For single cell voltage information, two different CAN ports are required. With inverter three CAN ports are then required.

So I'll create an example where only the PCS port is used and one with two CAN ports and one RS485 as server for the main yambms node with the inverter CAN port (as I would like to use it this way).

Maybe it is possible to use three CAN ports (1x internal, 2x external via SPI bus), but I've never tried this.

Back to the implementation, some information is not provided by the BMS, for example the starting balancing voltage, under voltage and over voltage limits. However each module has an energy counter and many other differences.

[image]

[image]

[image]

[image]

Is there a reason SoH is 99% when all batteries return 100%?

Sleeper85, 2025-03-29:
Good work, some unavailable information could be configured directly in YamBMS, as I've done for some BMSs like Seplos, but one of the most important pieces of information for YamBMS logic is `min_cell_v` and `max_cell_v`.

Regarding YamBMS's SOH, it is calculated based on the maximum number of cycles versus the number of cycles performed. This calculation should be moved to the BMS level so that it can rely on the BMS's SOH information when it is available.

GHswitt, 2025-03-30:
`min_cell_v` and `max_cell_v` is available. I made a draft pull request for review. I don't know the best way to include multiple module packages. For example:

bms_can_DEYE_PCS_Module.yaml and bms_can_DEYE_PCS_Module2.yaml are the same, except for the CAN bus IDs:

```
- can_id: 0x110
vs.
- can_id: 0x111
```

I would like to include the modules this way:

```
bms_can_pcs_module1: !include
    file: bms_can_DEYE_PCS_Module.yaml
    vars:
      pcs_module_id: '1' # must be a number
  bms_can_pcs_module2: !include
    file: bms_can_DEYE_PCS_Module.yaml
    vars:
      pcs_module_id: '2' # must be a number
```

Currently for the second module bms_can_DEYE_PCS_Module2.yaml is used.

Even better would be to make it completely dynamic as the main BMS reports the number of modules installed.

GHswitt, 2025-04-02:
@sloniu82: I've tested it today with my Deye SUN12K and the Deye battery. If you want you can give it a try:

https://github.com/GHswitt/esphome-yambms/tree/Deye_CAN_PCS

It is not the final version, the PR is still in work.

The file I use for my testing is multi-bms_DEYE-PCS_CAN.yaml
I recommend to first only connect the battery and check in Home Assistant if the values are looking good.

sloniu82, 2025-04-03:
Sure, I'll try, but I overslept in preparation and need a few days to order an additional CAN header.
I'll come back when I'll be ready with hardware.

GHswitt, 2025-04-09:
Sorry, I did not have the time to update all files to the latest structure. I'm using:
https://github.com/GHswitt/esphome-yambms/blob/Deye_CAN_PCS/YamBMS_LP_DEYE-PCS_CAN.yaml
with an external MCP2515 CAN module on the SPI bus:

[image]

These are the pins I use on the T-CAN485 board:

```
# MCP2515
  canbus_node_2: !include
    file: packages/board/board_options_itf_canbus_mcp2515.yaml
    vars:
      canbus_node_id: 'canbus_bms_1'
      canbus_can_id: '4'
      canbus_bitrate: '500kbps'
      spi_mosi_pin: '12'
      spi_miso_pin: '5'
      spi_clk_pin: '18'
      mcp2515_cs_pin: '32'
```

GHswitt, 2025-04-09:
AFAIK the ESP32 has only one internal CAN controller. Hence the need of the external MCP2515 CAN controller on a SPI bus.

The can_id is not that important I think, but I wanted to follow the source from Psynosaur.

sloniu82, 2025-04-09:
Ok, that is very interesting - didn't know about that limitation of esphome.
Not sure if I can use MCP2515 with SPI, because my board is already using SPI for Ethernet controller.
I could try I2C version maybe to be safe. 
But in both situations I need time to order new modules :(

Now, for test purposes, I've commented out YamBMS canbus definition for inverter communication.
So now, i have only one canbus which should speak with battery and present data in HA.
And again - no luck
[logs_yambms_run (1).txt](https://github.com/user-attachments/files/19667505/logs_yambms_run.1.txt)
All values are empty.
In the evening I'll try to verify and/or swap CAN L and CAN H wires, and rx/tx ping - mybe here I have a mistake

Another thing - which port on battery case you're using? Are you connected both to YamBMS and Inverter in the same time ?

GHswitt, 2025-04-09:
From the RJ45 connector 4 is H and 5 is L. H and L are connected to the CAN transceiver (TJA1050 for me) and the TJA1050 is connected to the ESP32 CAN TX/RX ports.

How long is your cable? Do you have the termination resistor enabled on your CAN transceiver board?

For testing I've only used one CAN port (Atom S3 + CAN transceiver board) and only connected the battery. 

With the T-CAN485 + MCP2515 I was able to test battery+ inverter like on the picture above.

Later I will use at least two boards and connect them via RS485 (one board main node to inverter, other one battery node).

sloniu82, 2025-04-10:
Hi @GHswitt, after all your informations provider I was able to successfully start CAN communication.
More mistakes ware made in my preparation:
- You ware absolutly right about limitation of one cansubs per esp. As I mentioned earlier, For test purposes I've deleted CAN definition for YasmBMS inverter communication and "Component canbus is marked FAILED" error stopped.
- Second mistake - my CAN cable was connected not to PCS port, but to IN/OUT port. I thought that these ports share the same capabilities. I don't have LAN splitter (to connect PCS port both inverter and YamBMS) - so I tried to be clever and use IN port instead.
- Third mistake is wrong CAN wiring. Documentation says that IN/OUT port CANH/L are pins 1 and 2. PCS port CAN ping are on 4/5 pins.
So now YamBMS communication is up and running:
[image]

Right now, we have plenty of parametrers readed from Deye BMS.
Is it possble to add (at least a few ) write capabilities for important paremeters, like:
- max/min discharge current
- start balance voltage (does this battery have an active balancer?)
- bulk voltage
- float voltage
- turning on/off charging 
As far as I know, there is no application (or any other possibility) to chage any of Deye BMS parameters - all these values come from factory.

Yesterday I ordered an additional MCP2515 module to communicate with the inverter. So I'lkl be able to test full YamBMS control and safty capabilities. In the meantime, I'll try to DIY a battery pack with JKBMS (in May maybe) and will see how they cooperate together.

GHswitt, 2025-04-10:
Great to hear that it can now decode the messages from your battery.
About the parameters, max charge and discharge current are included (for each module and the whole array):

[image]

[image]

Balance voltage unfortunately is not transmitted/decoded. But there are balance indicators for each cell and I see for my battery that it balances if I charge to 3.45V (55.2V bulk). The balance trigger difference seems to be 50mV. Below it does not balance.

The maximum charge voltage and discharge voltage is also transmitted by the battery:

[image]

But I find 3.65V too high and I use 3.45V for charging the battery.
Maybe it helps if you find the datasheet for your cells. There is a field for it:

[image]

sloniu82, 2025-04-11:
To be honest, because all these compability problems, I sold my RW-M6.1-B in mid march and just DIY two new batteries with JK BMS and parallel them.
So, to make test for thios project, I use my parents house battery (ai-w5.1), and it looks the use exacly the same model:
[image]

In my earlier post I asked is it possible to modify some BMS values when using CAN communication.
YamBMS integration with JKBMS provide access to control aloust all configuration parameters of BMS.
And this integration I can (for now, I hope) only read values presented by BMS.
It would be great to have ability to change values like RCV, float voltage, etc.

Now, back to this integration, I checked history of most interesting entities and I'm a bit worried because of cells inbalance in this battery:
[image]

When the battery is full (or almost full), the difference between the cells shoots into space and reaches a value of over 200 mV! 
In the seme time, my DIY EVE MB31 batteries never exeeds 15mV of cells inbalance.

Just wonder, if this is because Deye Inverter is not giving enough time for balancer?
And if I switch to YamBMS, it will be able to wait enough time to properly balance cells?

sloniu82, 2025-04-14:
Hi.
My battery have:
[image]
in practice it has been working since September 2024.
I see that youre are right, Deye Inverter requesting:
[image]
and this is 3.65V per cell.
But when I checked battery voltage history, I see that it never reaches so high voltage, it always stops  @ 56.6V:
[image]

So my quess is that one (or more) cells hits overvoltage protection before whole battery get saturated.
[image]
Unfortunately this BMS doesn't provide voltages of individual cells, so I cant find which cell is so weak.
Wonder If this should be covered by Deye warranty?

Today I've ordered 16 pcs of CATL314Ah + JK BMS - unfortunately, the expected delivery date is mid-May. 
So I need to wait a month to test integration of Deye battery with JK BMS controlled battery.

GHswitt, 2025-04-16:
133 cycles doesn't sound much. I don't know the targeted tolerance by Deye, but it seems quite high or the battery needs more time to balance. After some hours mine gets down to ~20-30mV, which is still much higher than the DIY packs.

Sleeper85, 2025-05-14:
Done with https://github.com/Sleeper85/esphome-yambms/pull/37

closed 2025-05-14 by Sleeper85