Sleeper85/esphome-yambms #36: pylontech support

tzapu, 2025-03-20:
hello and sorry this is a stupid question, but can i aggregate actual pylontech batteries with other BMSs? 
Only asking because it's not mentioned explicitly here https://github.com/Sleeper85/esphome-yambms/blob/main/documents/README/Supported_devices.md#supported-bms

i ve got a stack of 3 pylontechs and 1 eel battery with jk inverter bms, all 15S that I would like to aggregate 

cheers

Sleeper85, 2025-03-20:
This feature is not yet supported.

When it is, I'll respond via this issue.

Sleeper85, 2025-10-29:
@ergetie 

Here we're talking about integrating **Pylontech** batteries as inputs to YamBMS.

The **PYLON** output protocol has been in use for a long time.

Regarding your V-TAC battery, without knowing which BMS is being used, it's impossible to help you.
In any case, what's certain is that it's not a Pylontech battery.

mikollar, 2025-10-29:
Here is updated yamBMS repo with Pylontech files https://github.com/mikollar/esphome-yambms. 

I'm not very proud of my code, it still needs a lot of polishing. There are 4 modules hardcoded in the code. Also, the RS485 commands are hardcoded in a string. There is no CRC calculation there .... I took inspiration from this python library https://pypi.org/project/pylontech/

ergetie, 2025-10-29:
> [@ergetie](https://github.com/ergetie)
>
> Here we're talking about integrating **Pylontech** batteries as inputs to YamBMS.
>
> The **PYLON** output protocol has been in use for a long time.
>
> Regarding [...]

Oh! Sorry, I don't fully understand the difference then... You mean that I can potentially already now connect my two v-tac batteries + the JK-BMS battery to yambms?

I though it was enough to know the communication protocol of the V-TAC batteries (they are using PYLON low-voltage v2), but it seems like I have to open a v-tac box to see what bms they use?

Sorry, if this is the wrong place to ask these questions... let me know if i should ask somewhere else!

MrPabloUK, 2026-11-02:
FYI @Sleeper85, this will be revisited this month.

Sleeper85, 2026-11-02:
I discussed this with a friend who has developed a new component for Pylontech batteries.
I should have integrated it back in September, but some unforeseen issues came up.

Here is the GitHub repository that will be used: https://github.com/Dackara/esphome-pylontech

@Dackara still needs to make some modifications to it to ensure compatibility with YamBMS.

MrPabloUK, 2026-11-02:
Great news! I'll be able to do some testing when ready, now that I have time for development again.