Sleeper85/esphome-yambms #42: Web_server off

dolnet, 2025-04-12:
I would like to follow the suggestions:
# Please note that enabling this component will take up a lot of memory and may decrease
# stability and be the cause of reboot depending on the capabilities of the board used.
# web_server:
# port: 80
# log: false
# ota: false

Despite this form, the web server is active.
Board: T-Connect
What can I do to disable access to the YamBMS web?

Sleeper85, 2025-04-13:
The web server is not active by default.
I don't understand how it could be active if the lines above are commented out?

GHswitt, 2025-04-22:
I'm also using a T-Connect and do not see any webserver or port 80 open.

Sleeper85, 2025-04-23:
Probably a configuration problem in his YAML, I'm closing this issue.

closed 2025-04-23 by Sleeper85