Sleeper85/esphome-yambms #46: Lowering the requested charge voltage with Deye SUN-12K leads to battery discharging into grid

GHswitt, 2025-04-24:
Hi,

I've a SUN-12K and I'm using bulk voltages of 3,45V (55,2V) and for float 3,35V (53,6V). What I've seen now is that when the EOC/Float is reached, the new lower charge voltage send to the inverter results in the inverter discharging the battery into the grid (until the voltage is reached).
I'm not sure that is what I like or whats intended. Any ideas?

[image]

GHswitt, 2025-04-25:
I've increased yesterday the float voltage but still today when it reaches cutoff it discharges the battery immediately into the grid:

[image]

Also not all BMS have reached 3,45V per cell so they never reach the balancing trigger voltage this way. Or am I expecting something wrong?

Sleeper85, 2025-04-25:
This is a known behavior with Deye and other inverters.

If some of your BMSs don't reach 3.45V, you must calibrate them or use the inverter offset.

GHswitt, 2025-04-25:
Thanks for your answer. Two are a Deye BMS which unfortunately I can't calibrate. I think I'll offset the voltage a bit in some place, it is not much of a difference.

As for the Deye behaviour, it is not desirable for me to charge the battery full (~12:00-16:00 in the summer) and when it is full, sending several kWh with full 12kW power to the grid (panels are then not used anymore...). Latest time is about 18:00 when the battery starts discharging.

I wonder if a feature would be helpful that decreases the requested charging voltages slowly, following a timer or even the battery voltage itself as the cells go down to 3,35V.

And maybe in combination with charge current control, as the Deye battery itself does: When reaching full capacity it lowers the charging current gradually down.

So instead of directly lowering the requested charge voltage, maybe first disable charging (and/or sets the charge current to 0) to stop the charging process and then after some conditions (timer/voltage) it lowers the requested charge voltage?

Sleeper85, 2025-05-14:
I had already done some work on this last year, but it caused problems with other inverters.

Something to think about...

GHswitt, 2025-05-21:
I've added "Automatic Float Voltage" with can be enabled/disable by a switch. When enabled and it switches to float, it gradually lowers the requested charge voltage in small steps down the target float voltage. This avoids high discharge currents as the battery is allowed to settle more slowly.

[image]

[image]

I will create a PR when I'm happy with it and tested it for some days with my Deye.

GHswitt, 2025-06-13:
Can be closed as I'm now using the solution from PR https://github.com/Sleeper85/esphome-yambms/pull/53

closed 2025-06-13 by GHswitt