Sleeper85/esphome-yambms #52: ESP-32-C3 restarts and/or disconnects

ligius-, 2025-05-16:
I know that the BLE stack can cause problems but I'm only trying to feed data to HA without doing anything crazy. The configuration below connects to 2 JK-BMSs and provides the aggregate. If the logger section is set to:
```
logger:
  level: INFO
```
the board stays online for less than two minutes. If I connect the serial debugger in parallel to the OTA one, I can see that the board restarts. With the hardware_uart set to UART0 it seems to work fine for a longer time while the original setting USB_SERIAL_JTAG doesn't.

Even with this, the keepalive fails, possibly unrelated:  `[W][api.connection:177]: Home Assistant 2025.5.1 (192.z.z.z): Sending keepalive failed 55 time(s), will retry in 1000 ms` at which point HA loses track of the board.
Might be related: `[W][component:239]: Component esp32_ble took a long time for an operation (867 ms).`

```
logger:
  level: DEBUG # VERBOSE / DEBUG / INFO / WARN

ota:
  platform: esphome

api:
  reboot_timeout: 0s

packages:
  device_board: !include packages/board/board_ESP32-C3_DevKitM-1.yaml

  bms1: !include
    file: packages/bms/bms_combine_JK_BLE_minimal.yaml
    vars:
      bms_id: '1' # must be a number
      bms_name: 'JK-BMS 1'
      bms_ble_protocol_version: 'JK02_32S' # JK02_24S < hardware version 11.0 >= JK02_32S
      bms_ble_mac_address: '***'
      bms_cell_max_cycles: '6000.0'

  bms2: !include
    file: packages/bms/bms_combine_JK_BLE_minimal.yaml
    vars:
      bms_id: '2' # must be a number
      bms_name: 'JK-BMS 2'
      bms_ble_protocol_version: 'JK02_32S' # JK02_24S < hardware version 11.0 >= JK02_32S
      bms_ble_mac_address: '***'
      bms_cell_max_cycles: '6000.0'

  yambms: !include packages/yambms/yambms.yaml

substitutions:
  name: '' # Name that can be added as a second prefix after the friendly name, you can leave it blank.
  hostname: 'yambms' # This is the name of the node. It should always be unique in your ESPHome network.
  friendly_name : 'YamBMS Demo' # Prefix added in front of all entity names and also to classify them.
  yambms_id: 'yambms1'
  yambms_name: 'YamBMS 1'
  yambms_update_interval: '1s'
  yambms_input_number_mode: 'slider'
  yambms_battery_chemistry: '1' # 1-LFP | 2-Li-ion | 3-LTO
  yambms_cell_count: '16'
  yambms_bulk_v: '55.2'
battery)
  yambms_float_v: '56.0'
  yambms_rebulk_v: '52.8'
  yambms_eoc_timer: '30'
  yambms_cutoff_timer: '60'
  bms_update_interval: '3s'    # frequency at which BMS data is refreshed, going below '3s' can cause problems
  bms_combine_interval: '5s'   # frequency at which BMS data is combined in YamBMS
  bms_cutoff_timer: '50s'      # time during which the end of charge conditions must be respected (cut-off + cells not equalizing)
```

With the hardware_port set to UART0 the board keeps running but eventually loses connectivity to HA. HA reports "Connection reset by peer". For reference, the WiFi signal while testing is very good, the BLE signal is somewhat poor.

GHswitt, 2025-05-21:
I was also not able to successfully use the JK BMS with the BLE component. It was very unstable and for that reason I switched to RS485 which works fine. I tried different ESP32 boards (RAM). But I've also had similar issues when I tried the ESPhome Victron BLE implementation. With that I saw reboots from time to time when I tried to monitor more than one Victron Smartshunt.

ligius-, 2025-05-21:
I managed to get it mostly stable by reducing the number of sensors/entities, now down to 93 from around 200. Some have been removed completely, some have been set to `internal: true`.

Sleeper85, 2025-05-30:
Bluetooth (bluedroid) consumes a lot of DRAM and must share the same antenna as WiFi, it is less stable than UART / RS485 supervision. Note that there are users who supervise 2x JK-BMS + 1x SmartShunt with an AtomS3 without problems.

closed 2025-05-30 by Sleeper85