Sleeper85/esphome-yambms #65: Circular dependency error

andrewfraley, 2025-06-15:
Using ESPHome 2025.5.2 with Commit 39543e8:

`ERROR Circular dependency detected! Please run with -v option to see what functions failed to complete.`

Then if I run with -v I see an endless stream of

```
DEBUG Running to_code in esphome.components.template.sensor (num 42)
DEBUG Waiting for variable bms1_cycle_capacity_raw
DEBUG Running to_code in esphome.components.template.sensor (num 42)
DEBUG Waiting for variable bms1_cycle_capacity_raw
```

```
logger:
  level: INFO # VERBOSE / DEBUG / INFO / WARN
ota:
  platform: esphome
api:
  reboot_timeout: 0s
substitutions:
  name: "" # Name that can be added as a second prefix after the friendly name, you can leave it blank.
  hostname: "yambms" # This is the name of the node. It should always be unique in your ESPHome network.
  friendly_name: "YamBMS" # Prefix added in front of all entity names and also to classify them.
  yambms_id: "yambms1"
  yambms_name: "YamBMS 1"
  yambms_update_interval: "1s"
  yambms_input_number_mode: "box"
  yambms_battery_chemistry: "1" # 1-LFP | 2-Li-ion | 3-LTO
  yambms_cell_count: "16"
  yambms_bulk_v: "55.2"
  yambms_float_v: "53.6"
  yambms_rebulk_v: "52.8"
  yambms_eoc_timer: "30"
  yambms_cutoff_timer: "60"
  shunt_update_interval: "3s" # frequency at which Shunt data is refreshed
  shunt_combine_interval: "1s" # frequency at which Shunt data is combined in YamBMS
  bms_update_interval: "3s" # frequency at which BMS data is refreshed, going below '3s' can cause problems
  bms_combine_interval: "1s" # frequency at which BMS data is combined in YamBMS
  bms_cutoff_timer: "50s" # time during which the end of charge conditions must be respected (cut-off + cells not equalizing)
packages:
  device_board: !include packages/board/board_ESP32-S3_LilyGo-T-Connect.yaml
  uart_esp_1: !include packages/board/board_options_itf_uart_esp_1.yaml
  uart_esp_2: !include packages/board/board_options_itf_uart_esp_2.yaml
  uart_esp_3: !include packages/board/board_options_itf_uart_esp_3.yaml
  canbus_node_1: !include
    file: packages/board/board_options_itf_canbus_esp32_can.yaml
    vars:
      canbus_node_id: "canbus_inverter_1"
  shunt1: !include
    file: packages/shunt/shunt_combine_Victron_SmartShunt_BLE.yaml # shunt_combine_Junctek_KHF_UART.yaml
    vars:
      shunt_id: "1" # must be a number
      shunt_name: "vss1"
      shunt_mac_address: REDACTED
      shunt_encryption_key: REDACTED
  bms1: !include
    file: packages/bms/bms_combine_JBD_UART_full.yaml
    vars:
      bms_id: "1" # must be a number
      bms_name: "BMS 1"
      bms_uart_id: "uart_esp_2"
      bms_rx_timeout: "150ms"
      bms_max_charge_current: "100" # A. Used to calculate maximum charge current
      bms_max_discharge_current: "400" # A. Used to calculate maximum discharge current
      bms_cell_ovp: "3.650" # V. Used by 'Auto CCL' functions
      bms_cell_uvp: "2.800" # V. Used by 'Auto DCL' functions and to calculate maximum discharge voltage
      bms_balance_trigger_voltage: "0.015" # V. Used by 'Auto CVL' functions
      bms_cell_max_cycles: "2000.0"
  yambms: !include packages/yambms/yambms.yaml
  canbus1: !include
    file: packages/yambms/yambms_canbus.yaml
    vars:
      canbus_id: "canbus1"
      canbus_name: "CANBUS 1"
      canbus_node_id: "canbus_inverter_1"
      canbus_light_id: "esp_light"
      canbus_link_timer: "5s"
  device_debug: !include
    file: packages/base/device_debug_ESP32.yaml
    vars:
      debug_name: "Debug"
      debug_update_interval: "5s"
```

Sleeper85, 2025-06-15:
Confirmed, it's due to this part of the code but I don't know why.
I can compile by deleting it.

```YAML
  # +--------------------------------------+
  # | Charging Cycle Capacity              |
  # +--------------------------------------+
  - platform: template
    name: "${name} ${bms_name} total charging cycle capacity"
    id: bms${bms_id}_cycle_capacity
    update_interval: ${bms_update_interval}
    unit_of_measurement: Ah
    state_class: total
    icon: mdi:car-battery
    accuracy_decimals: 0
    filters:
      - or:
        - throttle: 10s
        - delta: 1
    lambda: |-
      float capacity_raw = id(bms${bms_id}_cycle_capacity_raw).state;
      if (isnan(capacity_raw)) capacity_raw = 0;
      return capacity_raw +
             id(bms${bms_id}_charging_cycles_offset).state *
             id(bms${bms_id}_battery_capacity).state;
```

Sleeper85, 2025-06-16:
Fixed with the addition of a new template sensor.