Sleeper85/esphome-yambms #84: EG4 LLV2/Wallmount Batteries

RAR, 2025-08-14:
Hey!

I've been working on getting my EG4 Wallmount batteries working with YamBMS so I can mix them with some some other JKBMS batteries.

While my additions are working and I can combine the JK and EG4, I have a little bit left to go:

https://github.com/RAR/esphome-yambms/blob/main/packages/bms/bms_sensors_EG4_LLV2_RS485_bms_full.yaml

mostly around errors:

https://github.com/RAR/esphome-yambms/blob/main/packages/bms/bms_errors_bitmask_EG4_LLV2_RS485_Modbus.yaml

My Questions are:

What should `bms${bms_id}_errors_bitmask` be?  a sensor, a number, etc..?

EG4 Split some off into a few different registers: errors and protection and warning - should I just do the basic errors with the bitmask conversion or pull from the other registers as well?

Awesome project.

Thanks!

RAR, 2025-08-15:
Got it!  I'll give it a shot tomorrow.

closed 2025-08-15 by RAR

Sleeper85, 2025-08-15:
You can post your YAML here, I will add it to YamBMS.

RAR, 2025-08-15:
Files are in this PR: https://github.com/Sleeper85/esphome-yambms/pull/85