Sleeper85/esphome-yambms #89: adding support for WaveShare ESP32-S3-RS485-CAN

dlarue, 2025-09-05:
Do I have it right that in order to test this board(https://www.waveshare.com/wiki/ESP32-S3-RS485-CAN) out I have to fork esphome-yambms and then create a new board file using the info obtained from their example WS_GPIO.h?  Maybe copy board_ESP32-S3_LilyGo-T-Connect-Pro.yaml 

WS_GPIO.h:
`#pragma once

#include <Arduino.h>     // Reference the ESP32 built-in serial port library

/*************************************************************  I/O  *************************************************************/
#define GPIO_PIN_CH1      47    // CH1 Control GPIO

#define TXD1              17    // The TXD of UART corresponds to GPIO   RS485
#define RXD1              18    //The RXD of UART1 corresponds to GPIO   RS485
#define TXD1EN            21

#define TXD2              15    // The TXD of TWAI corresponds to GPIO   CAN
#define RXD2              16    // The RXD of TWAI corresponds to GPIO   CAN

/*************************************************************  I/O  *************************************************************/
void digitalToggle(int pin);
`

changing this section of LilyGo-T-Connect-Pro.yaml to:
  board_name: "LilyGo T-Connect Pro"
  # Interfaces GPIOs
  # uart_esp_1: RS485
  tx_pin_1: '17'
  rx_pin_1: '18'
  # canbus_esp32_can: CAN
  tx_pin_2: '15'
  rx_pin_2: '16'

Not sure how to specify SoftSerial on GPIO pins 1 and 2 so that the Victron Smart Shunt can be connected to its Rx pin.

dlarue, 2025-09-05:
I created a new file and setup the UART for the Shunt on uart_esp1 but haven't worked out the EspSoftwareSerial issue yet.

https://github.com/dlarue/esphome-yambms/blob/446bb091b6245af017382508023c1b14d3e5249d/packages/board/board_ESP32-S3_WS-RS485-CAN.yaml

I did have a problem with the CAN device requiring a Light object be defined and I had to create a dummy light set to an unused pin. On the Waveshare module the LEDs are tied to the Tx and Rx pins so there is no direct pin control of an LED needed.

Sleeper85, 2025-09-09:
I don't understand your problem with `EspSoftwareSerial`?

The three UARTs of an ESP32 can be assigned to the GPIOs of your choice, and they remain hardware UARTs.

I created the new [ESP32-S3-RS485-CAN](https://github.com/Sleeper85/esphome-yambms/blob/main/packages/board/board_ESP32-S3_WS-RS485-CAN.yaml) board.

Best regards

dlarue, 2025-09-09:
The Waveshare ESP32-S3 RS485-CAN module is a custom product with an ESP32-S3 on it but it only exposes to the public nice wire screw terminals on one end for a CAN connection and an RS485 connection. The other side has a small 4 wire connector with a cable which brings out GPIO-1 and GPIO-2 pins along with GND and 3.3V.  So in order to connect up the UART interface of a Victron Smart Shunt we need exposed UART pins and GPIO1/GPIO2 are not hardware UART pins. Hence the interest in a SoftwareSerial interface.

[image]

dlarue, 2025-09-09:
I didn't think so because I didn't find anything in the code defining a softwareserial interface. It's too bad Waveshare didn't provide a way to jumper other IO pins to those connector pins. Not everyone is comfortable messing with hardware and that's what makes this module so nice and it already comes in a protective case...
There is an onboard 20pin header which would likely allow for hardware UART access but would also require cutting holes in the case to get the wires out. Again, not a tough task for those who do this kind of thing but the 'software-only' crowd would likely walk away from that.

[image]

closed 2025-09-10 by dlarue

Sleeper85, 2025-09-10:
@GHswitt 

I understand you ordered this board. There is no mention of PSRAM on Waveshare's website, let me know how it is when you test it.

dlarue, 2025-09-10:
I see it as an SP3485E chip.
https://www.micros.com.pl/mediaserver/UIMAX3485_sp_en_UMW_0001.pdf

[image]

dlarue, 2025-09-27:
@meteox79 did you see the conversation about why the Basen RS485 wasn't working? The RS485 chip on the Waveshare ESP32-s3-RS485-CAN is half-duplex so it needs the pin 21 handshake/direction control. Does your JK BMS support that and are you enabling it?

mateox79, 2025-09-27:
> @meteox79 did you see the conversation about why the Basen RS485 wasn't working? The RS485 chip on the Waveshare ESP32-s3-RS485-CAN is half-duplex so it needs the pin 21 handshake/direction control. Does your JK BMS support that and are you enabling it?

Can u make me that changes for JK BMS? So, i can test with all the possible options?

mateox79, 2025-09-27:
Bluetooth and Can is OK with jk PB and Deye
Rs485 would be better

GHswitt, 2025-09-27:
I think for JK BMS you need to use https://github.com/Sleeper85/esphome-yambms/blob/main/packages/bms/bms_combine_JK_RS485_Modbus_sniffer_talk_pin.yaml
and set sniffer_talk_pin.

```
# RS485 Sniffer
      - path: 'packages/bms/bms_combine_JK_RS485_Modbus_sniffer_talk_pin.yaml'
        vars:
          sniffer_id: 'sniffer1'
          sniffer_protocol: 'JK02_32S'
          sniffer_uart_id: 'uart_esp_1'
          sniffer_talk_pin: '21'
```

I'm annoyed that they really did not spent some cents more for a chip with autodirection support, then the device would be great.

mateox79, 2025-09-27:
> I think for JK BMS you need to use https://github.com/Sleeper85/esphome-yambms/blob/main/packages/bms/bms_combine_JK_RS485_Modbus_sniffer_talk_pin.yaml and set sniffer_talk_pin.
>
> ```
> # RS485 [...]

Thanks. 
I have compiled and upload. Works!!!
Rs485 and CAN work perfect in JK PB

Sleeper85, 2025-10-06:
Sorry, I've been really busy this past month.
Is everything okay now with this board ?
Can I close the ticket ?

mateox79, 2025-10-06:
Yes. All ok with the help of GHswitt with the talk pin

michu79, 2026-01-06:
I have just done it in version 1.6.0 and if it works incorrectly, it disconnects from the inverter after some time

[image]

[image]

I have automation that restarts esp32 if yambms soc = 0

[image]

GHswitt, 2026-02-19:
@jonasman You need to use the `flow_control_pin: 21`. The Waveshare does not have auto-flow control/auto direction.

jonasman, 2026-02-19:
> [@jonasman](https://github.com/jonasman) You need to use the `flow_control_pin: 21`. The Waveshare does not have auto-flow control/auto direction.

thanks!! it works!! 
now i got a Tesla Wall Connecter 3 with a fake neurio making load balancing. Im using a shelly to read the house consumption