Sleeper85/esphome-yambms #95: Flashing ESPHome using command line - getting CMAKE error

chaosnature, 2025-09-29:
i followed exactly these steps found here which is similar to mine: https://github.com/Sleeper85/esphome-yambms/blob/main/documents/README/Installation_procedure.md

Just wondered what your thoughts are on this.

I uninstalled Esphome, PIP and Python, and reinstalled all, now the latest version.

have you or anyone else tried the command-line flashing, and it worked?

hopefully the images below show the errors...

[image]

[image]

AI Suggestion:

[image]

chaosnature, 2025-09-29:
Worked on Linux - so windows has issues.

[image]

red light , what does this mean?
i ask bcos i usually see a purple light blinking, so i assume it means i have not connected to the BMS Bluetooth yet

also , a little off topic

i missed the discussion on shunt.....when i get shunt , what does it go between?

the BMS and te ESP 
and is it just the negative terminal of the batteries going to shunt?
so this mean instad of esp pointing to BMS MAC it will point to Shunt MAC?
and as i have 2 batteries, i can use one shunt for the 2 batteries as lng as the amperage is considered? 

I refer back to my schmatic:

[image]

chaosnature, 2025-09-30:
for Li-ion based on the below : 

[image]

What are the recommended values for:

[image]

And also, my cell count is 14, can't remember if I left the config as 16s in my previous

chaosnature, 2025-09-30:
and where did you move this config to:

web_server:
  port: 80
  log: false
  ota: false

Sleeper85, 2025-09-30:
> and where did you move this config to:
> 
> web_server: port: 80 log: false ota: false

Look in the board options section of the YAML

chaosnature, 2025-09-30:
> > and where did you move this config to:
> > web_server: port: 80 log: false ota: false
> 
> Look in the board options section of the YAML

/packages/yambms/yambms_web_server.yaml

chaosnature, 2025-09-30:
i don't want password when i access the webserver - what to comment out?

#ota:
#  - platform: web_server

web_server:
  port: 80
  version: 3
  local: true
  log: true

  # ota: false
 # auth:
#    username: !secret web_server_username
 #   password: !secret web_server_password

chaosnature, 2025-09-30:
Not sure what i am doing wrong but can't access the webserver over the configured IP

[image]

Logs : 

[image]

Ignore the not connected, i just have not disconnected the live ESP S3 yet.

chaosnature, 2025-09-30:
Fixed it nm...